aredden/electron-G14Control #4: Allow "profiles" with different options for each category, with power detection

carverhaines, 2020-12-13:
An option to create a profile that has a set fan curve, boost, and all the other settings.
Option to automatically switch plans based on whether on battery or A/C

aredden, 2020-12-13:
I'm actually working on that right now : ). The next feature is the grayed out menu item "Plans" which is exactly this.

aredden, 2020-12-15:
Alright, I just made another release with auto updates and g14control plans. It's possible it will be a bit buggy, but at least you won't have to keep checking for updates if it does end up breaking haha : ) @LavaTiger99 https://github.com/aredden/electron-G14Control/releases/tag/v0.1.27

LavaTiger99, 2020-12-16:
Awesome! Trying it out now.

LavaTiger99, 2020-12-16:
Seems to work great!

1) the last applied plan should automatically apply again on startup, because windows and asus may set things on boot, there may need to be a slight delay before applying the plan.

2) allowing power switching depending whether battery or A/C power is used.

LavaTiger99, 2020-12-16:
3. Current plan should show at the top of the Status page. If any one element is change outside a plan it should switch to "No Plan" or "Custom" or something like that.

LavaTiger99, 2020-12-16:
4. Ability to edit a plan after creating it.

LavaTiger99, 2020-12-16:
5. deleting all the plans crashes the program

LavaTiger99, 2020-12-16:
6. Getting some error when applying a plan, does not appear that the ryzen settings are applying

```
[RyzenADJ - ERROR] - Error setting ryzenadj settings: {"err":{"killed":false,"code":4294967295,"signal":null,"cmd":"cmd /c \"C:\\Users\\Carver Haines\\Documents\\dev\\electron-G14Control\\electron\\ryzenadj\\ryzenadj.exe\" --fast-limit 40 --slow-limit 35 --stapm-limit 25 --stapm-time 200 --slow-time 2 --tctl-temp 85 "},"stderr":""}  
```

aredden, 2020-12-16:
Yeah the plans page is still a work in progress. A lot of the things you mentioned are things that I plan to add. I think that I fixed some of the issues in version 0.1.28 such as crashing, although my testing was not thorough at all. The biggest thing I want to focus on right now is the auto-updates, since they are actually not working, although since it takes two full updates to actually test the updates (upload functioning version, then updating again), I will also work on these issues.

 The cpu tuning stuff does occasionally fail (RyzenADJ). It's just the way it is. I don't know why, but sometimes it'll take as many as 4 attempts to get it to apply.

closed 2020-12-24 by aredden