aredden/electron-G14Control #5: Allow changing screen refresh rate between 60 and 120 Hz (on applicable models) to save battery

LavaTiger99, 2020-12-16:
Originally we figured this out using ChangeScreenResolution.exe, but there might be a more elegant way:
https://github.com/CappyT/g14control/issues/1

Also allow this to be set in the plan.

LavaTiger99, 2020-12-16:
First polling the screen resolution options to see if the model has a 120hz option, then switching.

LavaTiger99, 2020-12-16:
Nevermind, I found it under the dGPU section. 
I think that should be it's own side menu, because the resolution changes apply whether the iGPU or dGPU are being used.

I would like to see integration of the refresh rates into plans still as well.

aredden, 2020-12-16:
That binary has been buggy for some people. I am tempted to remove it altogether, since you can just use windows to change refresh rate. Anyways, for many people changing the refresh rate for their laptop display doesn't actually do anything, even when changing using windows.

LavaTiger99, 2020-12-18:
I've been testing this quite a bit. I seem to have no problems switching between 60hz/1080p/32bit and 120hz/1080p/32bit, however I have noticed bugs and crashes when switching to any other resolution or bits.
I have also noticed (when idling on power saving mode) up to a 1000mW decrease in battery usage on 60Hz, and it does seem to accurately change the refresh rate according to Windows and NVIDIA control panel.
So perhaps making switching between 60hz/1080p/32bit and 120hz/1080p/32bit available as a toggle, then having the other options removed or hidden under an advanced/buggy/may not work disclaimer.
For this to replace the old G14 control app for me saving that extra battery saving on 60Hz is necessary.
Let me know what you think.

aredden, 2020-12-18:
Make sure it actually works. Go to https://www.testufo.com/ to see if it's truly 60hz or 120hz. Many people tell me that changing refresh rate doesn't work. You would instantly be able to tell without a doubt that it's 60hz because it's much choppier. Even when windows shows 60hz sometimes it's not actually the true refresh rate.

LavaTiger99, 2020-12-19:
Hmmm, it does look like it is not changing, even when changing with Windows or Nvidia... Strange. So I guess there is no way to adjust the refresh rate on the G14...

closed 2020-12-24 by aredden