aredden/electron-G14Control #8: Unify the menus

LavaTiger99, 2020-12-16:
The Top menu is obscure, and not needed since you have such a nice side menu.

I vote moving 'Logs' and 'Enable Start on Boot' to the settings side pane, and then maybe moving 'Exit App' to directly below the settings menu button (maybe in red text). Then eliminating the top menu all together.

aredden, 2020-12-16:
I kind of like it. It may be obscure but it makes it easy for people to open and send me logs, and gives an option to exit the app if you don't want to have to exit it from tray icon menu, without making it a button you accidentally click.

LavaTiger99, 2020-12-17:
Well maybe at least the Enable on Boot button to put it along with the ROG enable/map section? I created a startup task manually initially because I didn't think there was any start on boot option

knewknow, 2020-12-20:
Not sure about removing the top menu, but I didn't even know that it existed until I read this here. Was just about to setup a task to start on boot :). Thanks @aredden btw ... nice app. Any thoughts of putting the draw rate in the taskbar icon? Would remove the need for batterybar. Unfortunately we need to monitor this to know when to ResetGPU :).

aredden, 2020-12-20:
Yeah maybe I should make the top menu more pronounced. Thanks : ) I think it would be a bit difficult to add a taskbar-tray app like BatteryBar, but I could look into it.

closed 2021-01-12 by aredden