aredden/electron-G14Control #15: G14 Object Destroyed After Hibernate/Sleep

Spazzinq, 2020-12-21:
[image]

aredden, 2020-12-22:
What version are you using? Also, can you give me some details about what happened before this error occured?

Spazzinq, 2020-12-23:
0.1.31 (and previous versions but probably with different line #s)

I put my laptop into hibernation mode (I set my lid closing to hibernate it), and upon waking the UI doesn't appear, instead displaying this error.

aredden, 2021-01-05:
I tested this issue and it didn't happen for me. Does it still happen? Or is it an issue that is still occurring.

Spazzinq, 2021-01-05:
Looks like it doesn't for me anymore. Thanks for the fix!

closed 2021-01-05 by Spazzinq