aredden/electron-G14Control #19: missing features from G14Control-r2

IGonza, 2020-12-27:
I have G14Control-r2 installed on my g14 and I'm comparing this new version to the old one.
There are a couple features missing in this version:

1. When I plug/unplug power supply it automatically changes cpu boost, cpu_tdp, screen refresh rate and fan curve. 
2. There is no way to change screen refresh rate in this new gui at all.

Is it possible to add those 2 features? 
Thanks

aredden, 2020-12-27:
Changing refresh rate doesn't seem to actually do much, but you actually can change refresh rate on the "DiscreteGPU" Page. The refresh rate issues are talked about here: https://github.com/aredden/electron-G14Control/issues/5

As for 1., It's something I'm working on and will get to. It shouldn't be too hard.

IGonza, 2020-12-27:
About refresh rate:
Those selected checkboxes do not look like clickable, so I missed that part. I also do not see any difference on the website when change the rate from 120 to 60. But I'm not sure I can trust this because the browser has limited access to the system information, so probably it can read that display supports 120 but cannot read that it's running using 60hz at the moment..

aredden, 2020-12-27:
I have had it truly switch to 60hz, and it does show as 60hz on the website. The problem is that for some reason the refreshrate driver stuff is buggy on the g14.

jQubz, 2020-12-28:
I would also love to see these 2 options added. Refresh rate change works for me at least between 100p/60 and 1080p/120 in current version, but it would be very cool to be able to add it to a plan.

aredden, 2020-12-28:
Have you checked https://www.testufo.com/ to make sure it actually does change? @jQubz

jQubz, 2020-12-28:
Positive! Changed it to 1080p/60 through app, went on site, it said that my refresh rate is 60Hz. Then changed it again to 120Hz, site said 120Hz (and it was subjectively smoother).

aredden, 2020-12-29:
Oh cool. Well, for most people it doesn't work, so adding it to plan options would only work for a few people, and I have had people who had the screen res changer actually cause problems.

closed 2021-01-12 by aredden