aredden/electron-G14Control #23: Add custom CPU Tuning presets

jQubz, 2020-12-28:
I would love to see a possibility of making my own custom presets in CPU Tuning. Adding them to a plan would also be very nice.

aredden, 2021-01-02:
That would be cool, and I do want to do that personally, I'm just afraid that people will make custom plans that are dangerous. For example, before I had the minimum/maximum values set, my friend set his tdp to 1 watt and got a bsod lol.

jQubz, 2021-01-02:
Haha, that's good :D Well, you already allow users to change and apply not sane values tbh, but I totally get your point. Personally I would like to go even lower on power consumption to like 15W TDP (so basically 4800U instead of HS in my case). How about something like an advanced switch then? Or even better just hard block to some sane values like STAPM lower than 5W or more than 50W is probably not good and can be blocked (we would need to find info on these values for other parameters though)

aredden, 2021-01-07:
This update gives cpu tuning presets :)

closed 2021-01-07 by aredden