aredden/electron-G14Control #24: Fan Profile does not persist after booting from hibernated state

GreaseCrow, 2020-12-31:
Hey, just wanted to drop this incase the issue wasn't known.

Fan Profiles don't seem to persist after I boot my G14 from a hibernated state on Windows. I'm not sure if the rest of the profile is working either, but fans don't seem to be at the very least and require me to reapply the profile upon boot.

Thanks!

aredden, 2020-12-31:
Yeah that is normal. I am just worried that if there is an issue with my app, when you wake from sleep / hibernation / booting, then having the erroring plan reapply wouldn't be good.

GreaseCrow, 2020-12-31:
> Yeah that is normal. I am just worried that if there is an issue with my app, when you wake from sleep / hibernation / booting, then having the erroring plan reapply wouldn't be good.

I see, thank you for replying. Is it possible to add an option for it to persist with a warning to the user? My settings seem to be working well. Maybe an a hold key upon boot to prevent settings from applying on boot?

Thank you for your hardwork!

aredden, 2021-01-02:
I'll think about doing that. I want to wait and see if people have issues with the fans more, because I've had people who couldn't apply fan curves or even some AC fan profiles because of incorrect driver versions. I'm not sure that I'm confident enough for that feature right now :(

Filechaser, 2021-01-05:
I have the same problem. The Plan does get overwritten by the bios when starting from hibernation or from standby.
Can you change your software to reapply the plan on startup from hibernation and standby?

knewknow, 2021-01-10:
I ended up removing the application because I thought that it wasn’t working correctly/reliably (not applying on wake or reboot). I suspect that most users set their parameters and then never open the app again with the expectation that the modes will be persistent over reboots/sleep/hibernate.

aredden, 2021-01-10:
I apologize for trying to ensure the safety of your laptop? I mean you have to realize that me not implementing this feature is for your safety, not me being lazy or building a faulty application.

aredden, 2021-01-10:
Okay.. I have decided that I will implement G14Control plan on boot / resume / unlock-screen, since there haven't been any reports of damage as a result of auto switching plans.

Filechaser, 2021-01-10:
Even if....
1. You can just display a warning. 
2. The Bios has a temperature failsave that sets the fans to 100 percent or throttles the cpu.
3. If it is just a blue screen of death, then booting into windows save mode, will always be an option to disable your programm. 

Keep up the good work

knewknow, 2021-01-10:
@aredden sorry if I came off as criticizing without providing much detail or being constructive at all. I can see that now after re-reading (I typed it up quickly on my phone this morning). Anyway, I think that the work you've put in so far is absolutely fantastic and I appreciate all of the work you've put in and continue to put in on this project.

As a user, I do feel that your tool can solve many of the problems that have been reported on the subreddit. I just found that profiles not being persistent on reboot/sleep etc. was bringing about some additional uncertainty. When I was troubleshooting this, I wasn't sure if there was a bug in the tool, or a driver issue, or armory crate etc.... so I ended up removing and then re-testing it after every release hoping that it was a bug that was resolved. I only read this morning that it was by design (for good reason of course). I'm really happy that you're taking a 2nd look at this and potentially adding it into a future release :). Thanks for all you do!

GreaseCrow, 2021-01-10:
> 
> 
> Okay.. I have decided that I will implement G14Control plan on boot / resume / unlock-screen, since there haven't been any reports of damage as a result of auto switching plans.

Much appreciated! I think most of us aren't too afraid of damage or long-term "boot" related issues. I would suggest a configurable timeout or hold key to prevent boot settings from applying.

Thanks again for your work! Excited for the next update!

aredden, 2021-01-11:
I think apply on boot / resume / unlock screen should work now. I released the update. : )

closed 2021-01-12 by aredden