aredden/electron-G14Control #25: Auto Power Switching plans displayed are outdated

lmgarret, 2021-01-03:
Hello ! First of all, thanks for your hard work this great project!

Here an issue I have noticed with the 'Auto Power Switching' feature: if you have just created a new plan (in the same page), it won't appear in the dropdown list of plans to select from for both AC and DC plans. You'll need to restart the program to be able to select it

aredden, 2021-01-03:
Oh.. I think you can just switch to some other page, then switch back (for the time being), I don't think you need to quit and reopen. I will fix this asap :).

lmgarret, 2021-01-03:
You're right haha... anyway it's not super important as it's not the kind of setting you'll change every day :)

aredden, 2021-01-07:
This issue should be fixed, or will be fixed in the next update. (I don't remember)

closed 2021-01-07 by aredden