aredden/electron-G14Control #27: [suggestion] Switching plans from tray icon

lmgarret, 2021-01-03:
Currently we can change the Windows Plan, but I'd argue that it would now make more sense to change 'G14Control-v2 plans' from the tray icon's context menu instead?

LavaTiger99, 2021-01-07:
I second this, and would also like to add displaying current plan when hovering over or clicking icon.

aredden, 2021-01-07:
I'll get to it when I can, but it's not very high priority since there are other things I want to do first.

lmgarret, 2021-01-30:
I looked a bit into it and it'll be pretty difficult to implement, especially because [electron tray menu are not meant to be dynamic](https://github.com/electron/electron/issues/12633) :/

hofferkrisztian, 2021-05-25:
> I looked a bit into it and it'll be pretty difficult to implement, especially because [electron tray menu are not meant to be dynamic](https://github.com/electron/electron/issues/12633) :/

What about 3 static items in the tray, like 'G14C Plan#1', 'G14C Plan#2', 'G14C Plan#3' and you could map them in the GUI?