aredden/electron-G14Control #28: No CPU temperature and no battery drain value displayed

lmgarret, 2021-01-03:
As said in the title, both values are not read on my G14:

[image]

I'm not sure what could be the cause of it so here are some info:
 - Model: ROG Zephyrus G14 GA401IV_GA401IV
 - BIOS version: GA401IV.218
 - CPU: AMD Ryzen 7 4800HS
 - HWMonitor is also running (started at boot), is it possible that two processes cannot read the same sensors?

For my usage it really is a minor issue though, but might be nice to fix for others :)

aredden, 2021-01-03:
If you aren't using the english language setting, the regex I use to get the boost values from powercfg can't be parsed. Also, the temperature is from perfmon, and it suffers from the same issue. I do plan on eventually fixing this.

lmgarret, 2021-01-03:
Oh my bad, I remember reading about this now

MoonLightGO, 2021-01-27:
I'm also waiting for a fix for the Russian language

hbede, 2021-02-23:
I didn't know it was a feature, but it isn't working for me as well. My Windows is set to Hungarian. If there is anything I can provide for you or there's any way I can help let me know!