aredden/electron-G14Control #29: 0.1.36 - Change "Battery -> Enable Charge Modification" to a momentary button rather than a toggle

markfm62, 2021-01-04:
The way the ECM is working is that if there's a transition from False to True there's a one-time sending of the %.  If the user then changes the position to a new %, they need to turn ECM off (False) then back on again (True), for the new % to be used.

Either ECM should be made a momentary button (e.g., Save Charge %), clearly a one-off, else if ECM is True the SW should monitor for changes in the % slider position, and dynamically change the battery charging %.

aredden, 2021-01-05:
I just made a change where every time you move the slider (if enabled) it will submit a new battery limit, which I think solves the issues.

markfm62, 2021-01-05:
That would be fine, thanks for the rapid update!

closed 2021-01-07 by aredden