aredden/electron-G14Control #30: When applying a plan, the cpu preset doesnt get applied correctly

Filechaser, 2021-01-05:
[image]
It stayed the same when using the button "apply this plan" It should have used the Cpu plan "performance" and therefore a higher tdp

[image]

Filechaser, 2021-01-07:
ok thanks for adding custom cpu tuning profiles.
Yes even on the newest version, the tdp gets reset after boot from standby or hibernation and the plan doesn't get applied correctly

aredden, 2021-01-07:
Oh yeah, that would happen. When booting everything goes to the windows defaults. I havn't added plan application when you start the program, so that behavior would be expected. I'm more worried about plans not being correctly applied when you unplug / plug-in the laptop.

aredden, 2021-01-08:
Also, the UI is build with typescript, but using reactjs as a front end framework, redux for front end data storage, and it's built on top of electron-- which basically takes a website running on a single chrome tab and turns it into an app. @Filechaser

closed 2021-01-10 by aredden