aredden/electron-G14Control #31: Cannot create multiple G14Contol plans at a time

Alan-Cha, 2021-01-06:
When I try to add a second G14Control plan, I see the "Choose Plan Name" pop up, but when I click "OK", nothing happens.

My workaround is to click to a different tab, e.g. the Fan Curve Editor, and go back to the Plans tab.

I cannot seem to create multiple plans at once. Only one at a time.

***

Also thank you so much for this tool! This is honestly incredible work!

aredden, 2021-01-06:
That is strange. I'll look into it. :)

aredden, 2021-01-07:
I believe the issue that I solved with the *unable to add a plan when it has the name of a plan I just deleted* was directly related to this issue, and it should work when I release the next update.

LavaTiger99, 2021-01-07:
No, when you create and save one plan, the config settings for that plan are still selected, when trying to create another plan the save dialog just fails to save. You have to switch to a different menu (e.g. status) then back to plans to clear it and save a new plan.

aredden, 2021-01-07:
That's strange. It works for me with the current update. @LavaTiger99

aredden, 2021-01-07:
I believe this was an issue with the previous update, but it's fixed now. Can you confirm? @LavaTiger99

LavaTiger99, 2021-01-07:
Confirmed, it's fixed on the latest release

aredden, 2021-01-07:
Ok thanks! @LavaTiger99

closed 2021-01-07 by aredden