aredden/electron-G14Control #33: Fans Constantly ramping up and down, regardless of fan curve

MdotAmaan, 2021-01-07:
This was an issue i didn't have on G14control-r2, but when both cpu and gpu are at around 70C, the fans constantly ramp up and down, even if i set the fan curve to be completely flat.

aredden, 2021-01-07:
That is actually an issue with the laptop itself. It will happen in armoury crate (app) as well when you manually set a fan curve, as well as when you use atrofac. I generally only use custom fans when I know I'm going to be outputting a lot of heat. Also some curves cause oscillation more than others.

closed 2021-01-07 by aredden

Intenditore, 2022-03-08:
So there's no way to tame this and make laptop quieter?

MdotAmaan, 2022-03-08:
You could set the fans to turn off completely and then ramp up aggressivly under a heavy load. Thats how I've gotten around it, atleast for lighter use