aredden/electron-G14Control #39: CPU Tuning Revert Back To Default When Start The game

hrabdullah, 2021-01-11:
I just made my own presets in CPU Tuning but its turn back to default when I start a game. But its turn on again when I click apply again on G14 Control.

I made this presets to make stability between Power and Thermal output when boost are enable.
My G14 Configuration is R5/GTX1650TI.

Btw overall very good job. Sorry i didnt have credit card so i cant donate you :(
Sorry for my bad english. Thx.

[image]

[image]

[image]

aredden, 2021-01-11:
It's possible that the wattage it's showing in the beginning of a game is just the increased wattage average because of the post-boost wattage & the short term fastest boost wattage, so I'm not sure.

aredden, 2021-01-11:
You should also be aware that if you apply a fan curve, or an armoury crate plan, or a windows plan after setting the boost, it might overwrite the cpu tuning presets. I've also noticed that *sometimes* the boost values don't get applied when you apply a G14Control plan. I also think that you might actually get better thermals and fps if you just disable boost, since high temps like that will pretty severely cause the cpu to throttle hard.

closed 2021-01-14 by hrabdullah