aredden/electron-G14Control #40: custom fan curves has no effect

RodSmet, 2021-01-12:
Just updated to version 0.2.0 and tried to make plan with the minimum cpu fan speed at 15% but they remain at 0%. If I switch to one of the premade fan curves they ramp up but when I switch back to the custom they goes back down to 0%

aredden, 2021-01-12:
That is probably a result of a possibly defective fan controller in all G14's. The fan control software is all atrofac-cli , so there's not much I can do with regards to that. Here is a [thread](https://github.com/cronosun/atrofac/issues/5) where other people are having similar issues with [atrofac](https://github.com/cronosun/atrofac). The software communicates directly with the asus driver that is in control of fan speed. It's the same type of control that you get when you use the ASUS Armoury Crate (the actual asus software) custom fan curves. I'm not sure if there's really anything that can be done to make it better.

RodSmet, 2021-01-12:
I previously experienced that exact problem with custom fan curves. Now the changes don't seem to apply. The fan curve I made is based on the slow fan curve but instead of min at 0% I raised it to 15%. I tried stress testing it with Aida64 and it acts exactly as the slow fan curve. I however have no issues with the omega saver plan. Isn't that also a custom plan in the same sense? or is there more behind the scenes-magic going on with that?

aredden, 2021-01-12:
For me personally, I just use the armoury crate plans (in g14control) without using a custom fan curve, since those plans actually already have a default low speed fan curve that doesn't cause any oscillation like custom fan curves do. Another thing I noticed is that custom fan curves tend to only really like to work above about 20-25%, and when each successive thermal breakpoint has an increased percentage, such as 21%, 23%, 28%, etc... I personally ONLY use custom fan curves when I want serious cooling and fast fan speeds.

RodSmet, 2021-01-12:
Hm, but there is no armoury crate plan where the fans stops completely right?

aredden, 2021-01-12:
No. Armoury Crate plans have fan curves, but they are slower than what you can achieve with custom fan curves, unless they are similar to the slow fan custom fan curve.