aredden/electron-G14Control #42: CPU tuning does not switch to a custom cpu profile when switching power plans

Mirikov, 2021-01-15:
I have been messing around with the software for a bit in order to make some perfect plans for me. So I ended up creating 5 power profiles for different scenarios. And deliberately for the power saving plan I have set up custom cpu tuning profile. 
Fastest Boost TDP = 15
Post-Boost TDP = 10
TDP = 8
Temperature limit = 70
This is done to limit the CPU and to make it run silent with decent thermals.
 
However when I set my plan to saver mode it does not switch the CPU profile and defaults to numbers that are not even part of any of the profiles.
Fastest Boost TDP = 50
Post-Boost TDP = 35
TDP = 25
Temperature limit = 90

PS. Sorry for my English

aredden, 2021-01-16:
I know about this issue. I made it so that the last thing applied is always ryzenadj, and it still does not apply the changes. I am thinking it has something to do with a delay between some other choice (windows plan, fan curve, armoury crate plan) and the application of the ryzenadj.. so that ryzenadj "looks" like it's getting applied last, but some other setting is overwriting it after it gets applied regardless. The only think I can think of would be to have some longer delay between other settings and ryzenadj getting applied. I'll do some testing.

aredden, 2021-01-16:
Also, as to why it the numbers you are getting don't relate to any ryzenadj plan you have used before, I would recommend reading the info popup on the Fan Curve page ;). Okay fine haha.. The silent Armoury Crate plan will throttle CPU Tuning presets regardless of what they are by about 20%.

aredden, 2021-01-17:
@Mirikov The most recent update should fix your issue. Let me know if it works :)

Mirikov, 2021-01-17:
Ok it works like a charm now. Sorry that I didn't reply. Just have an exam coming up.

closed 2021-01-17 by Mirikov