aredden/electron-G14Control #43: Fan Control when using Armory Crate from within your software - buggy

Filechaser, 2021-01-18:
When using the Armory Crate option from within your app and a Fan Profile too, then the Fan Profile doesn't always get applied correctly.

Can you make sure, the software first switches the armory crate profil and then the rest. Then Probably check 30 seconds later if the fan profile really corresponds to the active profile. If not apply again?
Or as workaround make the plans visible in the context menu to fast apply them

aredden, 2021-01-18:
In order for an armoury crate profile to get applied with a fan curve, it must be at the same time. Also, fan curves are weird in general and frequently cause fan oscillation. This isn't because of my app.. it's because of the g14 hardware.

ProNoob404, 2021-01-28:
@aredden is it possile to set fan curve with minimal rpm but not fanless? I used to try with silence/performance and slow fan. Everytime when I set something above 0% rpm It always droped to 0, doesnt matter 20 or 50%