aredden/electron-G14Control #47: Remaping ROG key to your software doest work correctly.

Mirikov, 2021-02-15:
Hello there, me again. 
I have a fresh windows install without the armoury crate installed. I have disabled rog and remapped the key from your software but every now and then I have a pop up when I press rog button saing that I need to install armoury crate. The only way I found to remove it is to enable armoury and to disable it again it maked it go away but not for long. Also this may have to do with it but sometimes the remap glitches and then press rog key the g14control pops up and instantly hides to tray. If I undo the remap the rog key makes g14 control pop up. 

PS sorry for my English.

aredden, 2021-02-15:
Aahhh that's an interesting edge case. I didn't think about that. I guess I figured that since you need ASUS System Control Interface V2 for fan curves to work that people would probably have armoury crate installed. I'll try to fix this soon.

Mirikov, 2021-02-15:
> 
> 
> Aahhh that's an interesting edge case. I didn't think about that. I guess I figured that since you need ASUS System Control Interface V2 for fan curves to work that people would probably have armoury crate installed. I'll try to fix this soon.
I do have System control interface but they are provided as a seperate package as well as certain software for the anime matrix.

aredden, 2021-02-15:
Armoury Crate actually provides two windows plans, which I use in the default config for the app. It provides a "High performance" plan and a "Power saver" plan, which are used when you use the *silent* armoury crate plan (Power saver) or the *performance* / *turbo* armoury crate plan (High performance) plans.
Basically...

| **Windows plan** | **Armoury Crate** |
|-------------------| ---------------------|
|  High Performance  |  Turbo or Performance  |
|  Power Saver  |  Silent  |

Mirikov, 2021-02-16:
Oooohh, ok. Honestly I deleted the original presets and set up my own. Like for perfomance I ramp up the fans to 100% all the time so that g14 can sustain good boost numbers and cool temperatures. For the power saver I use slow fan profile and limit the tdp of the cpu to like 6 watts. Those settings recetly were able to give me around 14 hours of light work. Also I did not do anything to windows it self yet in order to improve battry.    
That is something I wasn't able to achive with armoury crate which renders it useless thats why I dont even bother installing it.

hbede, 2021-02-23:
Hi!
I've been using your app for months, but since the last update, the ROG button started to reset itself after every reboot. It opens G14Control just fine, but it also opens a window suggesting to download Armoury Crate. Mine PC is a fresh install, but I installed almost every Asus apps from MyASUS. After a few months I found your app and I thought it was amazing! I downloaded it, and after a few days I decided to delete Armoury Crate. It'd been fine since then, except the last few days (or weeks?).
I can fix the problem temporarly by enabling and disabling ROG, but it resets after every reboot.