aredden/electron-G14Control #55: Plan failed

exedox, 2021-04-02:
I'm attempting to apply a plan, but it fails.

[image]
I'm not sure how to fix this.  Is there something blocking, or is this a feature that's not available yet.  I just did a clean install of windows.

shader, 2021-04-02:
I'm having the same problem; maybe a recent windows update that blocks it? It used to work.

Ik00k0, 2021-04-13:
also having this problem

M3gaFr3ak, 2021-04-14:
This error is reproducible for me as well. A custom plan applies perfectly while the builtin plans don't. Maybe some of the selected options in the builtin plans contain settings that don't work on the system? E.g. nonexistent Windows power-plans?

aredden, 2021-04-14:
Yeah the plans that I provided have windows plans that are specific to Armoury Crate. If you have uninstalled Armoury Crate then you won't have those plans, since Armoury Crate actually removes the windows plans when it gets uninstalled. The Windows plans are "High Performance" & "Power Saver". Check the app logs to see what errors you guys are getting. e.g. -> top-menu -> open logs.

aredden, 2021-04-14:
Also, logs don't actually get written if it's opened on start, so if you want to see the actual error you are getting, close it and reopen, then it should start writing to a log file in the directory where "Open Logs" directs you.

exedox, 2021-04-14:
That's kind of the odd thing, I had Armoury Crate installed because I read that it needs to be there for the plans to work.  However, with it installed, the plans still didn't work.  I'll post the error log soon.

M3gaFr3ak, 2021-04-14:
> 
> 
> That's kind of the odd thing, I had Armoury Crate installed because I read that it needs to be there for the plans to work. However, with it installed, the plans still didn't work. I'll post the error log soon.

reinstalling armoury crate didn't bring back the power plans for me, I guess they only appear on a fresh asus-install (over recovery) or so.

exedox, 2021-04-14:
> reinstalling armoury crate didn't bring back the power plans for me, I guess they only appear on a fresh asus-install (over recovery) or so.

I was on a fresh install when I did this.

[Powercfg - ERROR] - Problem setting boost ac power index for guid: 64a64f24-65b9-4b56-befd-5ec1eaced9b3 and boost 0. Error:
{"err":{"killed":false,"code":1,"signal":null,"cmd":"powercfg /setacvalueindex 64a64f24-65b9-4b56-befd-5ec1eaced9b3 SUB_PROCESSOR PERFBOOSTMODE 0"},"stderr":"The power scheme, subgroup or setting specified does not exist.\r\n"}
[Powercfg - ERROR] - Problem setting boost dc power index for guid: 64a64f24-65b9-4b56-befd-5ec1eaced9b3 and boost 0. Error:
{"err":{"killed":false,"code":1,"signal":null,"cmd":"powercfg /setdcvalueindex 64a64f24-65b9-4b56-befd-5ec1eaced9b3 SUB_PROCESSOR PERFBOOSTMODE 0"},"stderr":"The power scheme, subgroup or setting specified does not exist.\r\n"}
[HIDControl - INFO] - Found file ArmouryCrate.UserSessionHelper.exe_disabled at path: "C:/Program Files/ASUS/ARMOURY CRATE Service"
[GPUScripts - INFO] - Error setting AC value for switchable dynamic graphics...
PS_CMD_FAIL_ERROR: 
      The power scheme, subgroup or setting specified does not exist.

[G14ControlPlans - INFO] - There was an issue setting graphics to: 0
[G14ControlPlans - ERROR] - Couldn't set boost & graphics preferences.
[G14ControlPlans - ERROR] - Failed to set G14Control plan.

aredden, 2021-04-14:
That's exactly what I thought. You don't have the default Armoury Crate windows plans, so those default power plans won't work.

exedox, 2021-04-14:
> That's exactly what I thought. You don't have the default Armoury Crate windows plans, so those default power plans won't work.
Ok, well in any case, it's no big deal.  I wanted to see what the issue was,  I recall also using the debloater to remove armoury crate now.  I can still do manually done plans, which provides same result!  So if anyone runs into same issue at least this is here to answer the issues.

shader, 2021-04-19:
Mine started working again after another update I think, though not reliably. Sometimes it doesn't work, but now a reboot can fix it.

wintea, 2021-05-16:
I reinstalled mine and after that it started working again.

DralrinResthal, 2021-07-16:
It may now be July 16th but I just had a fresh installation of Windows after an idiot thought of my own.
Anyways, I have tried both with and without Armoury crate and I cannot apply any plans, custom or default.

imjustanoob, 2021-08-08:
Figured out a workaround for this. 

open a CMD prompt and run 
`powercfg /L`

Then export your "Power Saver profile" If you don't have one, go create one.
`powercfg /export c:\profile.pow {GUID}`

To make things clean I went and deleted my current power saving profile in power options. Or you can just rename it and import the one you just exported. I'm not sure if the GUID it is looking for is the same for everyone. But mine and the one above are the same.
`powercfg /import c:\profile.pow 64a64f24-65b9-4b56-befd-5ec1eaced9b3`

open g14control and apply settings now.