aredden/electron-G14Control #59: dGPU enable/disable

flukejones, 2021-05-04:
Hi! Maintainer of https://gitlab.com/asus-linux here. Been wondering how exactly you enable/disable the dGPU on Windows.

I see ` SW_DYNAMC_GRAPHICS = "e276e160-7cb0-43c6-b20b-73f5dce39954"` and related calls using it, but I'm having a hard time mapping out precisely what that call is.

Through a whole lot of persistence I've written this [kernel patch](https://gitlab.com/asus-linux/fedora-kernel/-/blob/rog/0001-asus-wmi-Add-dgpu-disable-method.patch) to enable/disable the dGPU using the nasty ACPI hack ASUS seems to have included (this does in ACPI what I do in software - remove the dGPU from the device tree).

So I'm wondering if that call you do in your project maps to that. And I *think* it's recent models only, like the GA401QM.

I've solved most issues and reversed most things on these laptops, but there are still some small things like that.

aredden, 2021-05-05:
It's not actually a full enable / disable. I've found that fully disabling the gpu actually increases the power draw, which is frustrating. That is just a mapping to the registry entry for switchable dynamic graphics.. which allows you to change between 'power saving graphics' and 'performance', which I assume has to do with nvidia gpu vs igpu, since that is what windows describes when you change the default graphics renderer for a specific program.. 
[image]

aredden, 2021-05-05:
In other words, I'm guessing lol