aredden/electron-G14Control #63: Remap ROG button isn't working

kjaonline, 2021-07-01:
I get an Error Remapping ROG Key. Wondering if there's anything I can provide to help you debug.

Thanks!

sjjlee, 2021-08-16:
I'm having the same issue

kjaonline, 2021-08-16:
Yeah. Repo seems abandoned so I'm considering forking.

aredden, 2021-08-16:
I've been pretty busy with work so yeah, I havn't had time to work on personal projects like this. If there was an update to the keyboard driver then it's likely that the remap will be listening for the wrong HID event. I did however just get the cpu monitoring tool to work with cezanne (zen 3 aka g14 2021), since I've had a small break from work. I might end up fixing some of the issues with this project if I have enough time, so who knows.