aredden/electron-G14Control #66: Windows 11 issue

Fjallren, 2021-07-11:
All users who are running Windows 11 are experiencing issues where all plans, in-built and custom are not succesfully applying.
I am assuming the app requires to be updated for win11 compatibility. 

[image]

HemangKumawat, 2021-07-15:
just in case I can help. I am not having that issue.

Fjallren, 2021-07-19:
Idk if you can really help, since I wasn’t the only reporting this in the discord a few weeks ago

aredden, 2021-07-26:
Yeah the software is really designed specifically for windows 10 g14 from 2020. It doesn't have updates for the g14 2021, nor does it have updates for windows 11. I wouldn't be surprised if it doesn't work. Unfortunately I don't have time to fix it at this point since I got a job. :(

Fjallren, 2021-07-28:
Feelsbadman. I’m guessing we’re going to have to go for armoury crate and ryzen controller with win11?

closed 2021-07-29 by Fjallren

moracabanas, 2021-08-05:
> Feelsbadman. I’m guessing we’re going to have to go for armoury crate and ryzen controller with win11?

I used it mannually with windows 11 mainly for reset GPU, tweaking CPU power with some of their tools broken I.E temps are not read, and you cannot modify boost like _efficient aggresive_.

I updated windows 11 to 10.0.22000.120 build today and you cannot open it anymore, it just don't run.

RIP G14Control and thanks for all the work put into it @aredden ✌️

HemangKumawat, 2021-08-06:
That's so, weird. Because every single feature has been working for me even on windows 11 through day 1 and through every single update. I have now switched to the windows 11 beta tho. And yes still works. With all the auto switching of power plans and display of temp. Reading.

moracabanas, 2021-08-08:
> That's so, weird. Because every single feature has been working for me even on windows 11 through day 1 and through every single update. I have now switched to the windows 11 beta tho. And yes still works. With all the auto switching of power plans and display of temp. Reading.

Yeah, it is weird. It worked out once I tested the unnoficial 11 release, but now I fresh installed the official 11 preview and there is no luck. 

Did you fresh installed the upcoming beta or just updated from windows 10 to the "pre-preview"? I mean the leaked first one, that one worked with G14control.
Otherwise can you provide your Windows 11 build number?

Thanks you so much! 

I fresh installed and then I installed Asus armory crate with basic asus drivers to get it working, I also installed auto AMD drivers and GeForce ones.