aredden/electron-G14Control #71: Not an issue, just saying "Hi" :)

ioerrorle, 2021-09-30:
Hey man!
I tried your software and decided to go full try-hard. My goal is to disable all ASUS services that come with laptop and make a lightweight app that would replace even Asus Optimization service.
For now I made a piece that successfully reads current TDPs, fan speeds, APU temp and subscribes to ACPI events, but, most importantly, I successfully remapped Fn+Up/Down to Fn+C/V and made Fn+Arrow keys work as PgUp/PgDown/Home/End.
If you're interested in collab, let me know.
Project i'm working on:
https://github.com/ioerrorle/G14ControlPP
It's not even half-done, I'm spending most of my free time with IDA and x64dbg to understand what ASUS software is doing. You're welcome to participate.
It's a pretty hardcore challenge for someone who never touched C++/win32, but exciting.
Good job on a project anyways!

aredden, 2021-10-01:
Looks pretty cool! I hope it goes well 😄 . I'm not much of a c / c++ guy though, been more into rust recently. I don't know if I would ever dare to figure out how the ASUS drivers _actually_ work haha. Good luck!

khanh-ne, 2021-10-16:
Nice project man. The only useful function in Armoury Create is AI noise canceling,  and it works pretty well. I am looking for a way to use it without A.C. app from Asus.