aredden/electron-G14Control #80: App can't start

chuan0418-com, 2022-01-09:
[image]
I can't start the app, it pops up this error.
I'm using windows 11, and it actually work fine before days.

aredden, 2022-01-11:
Did you start with admin permissions?

chuan0418-com, 2022-01-16:
> Did you start with admin permissions?

Yes

aredden, 2022-01-21:
I'm thinking it might have to be something like.. the "Allow local powershell scripts to run without signing" windows setting?

chuan0418-com, 2022-01-23:
`        Scope ExecutionPolicy
        ----- ---------------
MachinePolicy       Undefined
   UserPolicy       Undefined
      Process       Undefined
  CurrentUser    RemoteSigned
 LocalMachine    Unrestricted`

I enter `Get-ExecutionPolicy -List` and it show up these setting.
But the app still show up the error. Is my setting correct?

aredden, 2022-01-25:
Well I know it's some sort of permission issue, since that's what an EPERM error is. It's probably related to powershell. Do you not have powershell installed? This is the file in question: https://github.com/aredden/electron-G14Control/blob/main/electron/src/IPCEvents/ComputerInfo.ts ..

aminbahjati, 2022-02-01:
[image]
[image]

I faced this problem today and these two errors raised.
What should I do?