aredden/electron-G14Control #86: Feature Request: Add ROG Key remap support for 2021 ROG G14 GA401QH

pratyakshm, 2022-02-14:
I understand that this app's readme mentions that it is only available for 2020 G14 devices, however I am using it on my 2021 G14 because I haven't come across such a useful and/or similar utility for 2021 G14 yet. Most functions in this program are working fine with my 2021 G14, but the ROG key remapper does not work.

Is there any way that you could add support for it? I can provide with whatever that's required for it.

langmeister, 2022-03-02:
+1 - works perfectly with my G14 2021 with the exception of this button remapping.  

I would very much like this and will offer up what I can (although caveat: I'm not a developer).

Model / BIOS Version is GA401QM.410

Vital-Fernandez, 2022-07-04:
Did you find a solution?

langmeister, 2022-07-04:
> Did you find a solution?

Nope.